nograx/ioBroker.zendure-solarflow #10: IObroker is not bild Objekte

Wunni73, 2023-12-17:
**[image]** Ordner ist lehr.
 bei **Instanzen** fehlt >Verbunden mit Gerät oder Dienst<

Wunni73, 2023-12-17:
[image]

Wunni73, 2024-01-15:
leider wurde der Ordner immer noch nicht gefüllt :-(
trotz neu aufsetzen des IOBroker und Neu Installation von Version 1.0.5

nograx, 2024-01-15:
Hi, ist dein Hub auf dem globalen Server registriert?

Wunni73, 2024-01-16:
mqtt.zen-iot.com  mit der MQTT abfrage geht es 
auf den Euro glaube ich 
[image]

Wunni73, 2024-01-16:
[image]

Wunni73, 2024-01-16:
kann ich mich bei beiden anmelden

nograx, 2024-01-16:
Nein das ist nicht möglich. Der HUB kann nur auf einem Server registriert werden.

Wunni73, 2024-01-16:
:-( ok, danke für die info

Wunni73, 2024-01-16:
ich nochmal
mqtt.zen-iot.com          damit frag ich momentan ab. geht 👍 

und das sind deine 2 Server 

 mqttUrl: "mq.zen-iot.com",                           #Global
 mqttUrl: "mqtteu.zen-iot.com",                    #EU

gibt es da einen unterschied ?

nograx, 2024-01-16:
mqtt.zen-iot.com ist der Dev-Server, bietet aber nur lesenden Zugriff. Die anderen beiden sind die offiziellen Server die die App nutzen.