nograx/ioBroker.zendure-solarflow #17: Akku Temperatur auslesen

SlipKoRn88, 2024-01-16:
Hallo,

bietet die API die Möglichkeit die Temperatur der Akkus auszulesen? Falls ja, wäre ich stark daran interessiert, diese Werte auch im ioBroker zu haben, damit ich meine separate Heizung für die AB1000 steuern kann.

Besten Dank

Wunni73, 2024-01-16:
Super Ide 👍

SlipKoRn88, 2024-01-16:
Das ist meiner Meinung nach nur die höchste Temperatur, die der Akku erlebt hat in Celsius. 
Sollte die Angabe wirklich in Kelvin sein, so wären die 6,95 Kelvin dann -266 Grad Celsius 

Bei mir steht bei maxTemp seit Wochen unverändert 26,6.

nograx, 2024-01-16:
Bitte mal die Version aktualisieren (1.0.7 ist per npm verfügbar, im beta kanal ab morgen früh). Die Info mit Kelvin stimmt zwar, ich habe in den ersten Versionen aber nicht in Celsius umgerechnet bzw. war da auch noch ein Fehler drin an den ich mich jetzt nicht mehr erinnere. Bei mir wird der Wert laufend aktualisiert und passt auch - siehe oben 6,95° Celsius. Die Umgebungstemperatur liegt aktuell bei 5° Celsius.

SlipKoRn88, 2024-01-17:
Ich habe die aktuelle Version drauf und werde es mir einmal anschauen! Besten Dank