nograx/ioBroker.zendure-solarflow #19: Funktionserweiterung

Forestarius, 2024-01-19:
Hi, ich schon wieder
Wäre es möglich den Bypass über den Adapter zu steuern. Dann könnte man sein Programm z.b. so anpassen , dass man erst ab einer bestimmten Eingangsleistung den Bypass auf Automatik nimmt.

Anscheinend scheint der SF immer auf Batterie first zu laufen.

Gruß Frank

nograx, 2024-01-19:
Ich werde mal schauen ob sich das umsetzen lässt. Den Parameter scheint auf jeden Fall steuerbar zu sein. 

Ich möchte an dieser Stelle eine Warnung aussprechen: Wie auch Zendure in der App angibt sollte man das Bypass Relais nicht zu häufig ansteuern. Dies ist vermutlich die erste Komponente die irgendwann im Hub kaputt gehen kann, und je häufiger dieses Relais schaltet desdo eher geht es kaputt!

Forestarius, 2024-01-19:
Okay, ist dies wirklich als Relais ausgeführt. Dann ist das Feature vielleicht Kontraproduktiv.

nograx, 2024-01-19:
Ja, ich gehe davon aus das die Akkus dann DC seitig getrennt bzw. zugeschaltet werden. Da fließen dann beim Schalten natürlich ordentlich Ströme und ich denke das dann entsprechend auch die Kontakte leiden. Nur so macht die Warnung von Zendure auch Sinn.

closed 2024-03-07 by Forestarius

nograx, 2024-03-22:
Hi, es ist mit der neusten Version 1.2.7 (heute per ppm oder GitHub, morgen auch per Latest Repo) auch möglich den Bypass zu steuern, exakt so wie es in der App einstellbar ist. Gruß, Peter