nograx/ioBroker.zendure-solarflow #20: activate the adapter - no values

markushoedl, 2024-01-21:
**Describe the bug**  
I activated the adapter - there is just the event: "Getting device list from Zendure Rest API!"
But no devices.

**To Reproduce**  
Tried several times. On the App there is an logoff - all events in iobroker show ok - but no values

**Expected behavior**  
see some values under the folder "zendure solarflow"

**Screenshots & Logfiles**  
[image]
[image]

**Versions:**  
 - Adapter version: v1.0.7
 - JS-Controller version: < v18.16.0
 - Node version: 5.0.16
 - Operating system: Docker

nograx, 2024-01-21:
Die you natives that your SolarFlow Hub has to registered in the global Server?

markushoedl, 2024-01-21:
ok now I get some values. some are not updating

[image]

nograx, 2024-01-21:
Don‘t use the App if you wish to use the Adapter. You will be logged out if you start the App. It‘s possible to use a second account. States will ne Updated as soon as a new value will be sent from the Hub.

closed 2024-02-29 by nograx