nograx/ioBroker.zendure-solarflow #21: Kein Objektpfad wird angelegt

Torham2, 2024-01-23:
Hallo.

Bei meinem ersten Versuch den eine Instanz des Adapters anzulegen ist mein Iobroker komplett abgeschmiert. 
Neu hoch gebootet, ging wieder.
Allerdings kann ich jetzt soviele Instanzen des Adapters anlegen wie ich will, es werden bei Objekten keine Pfade sichtbar. Das heißt ich habe unter Objekte zwar den Ordner Zendure-Solareflow, aber darunter keine Unterodner 0,1,2 etc für die Instanzen.

Auch ein löschen der Instanz und des Adapters und Neuinstallation des adapters und Instanzen brachte keine Veränderung. Sowohl die Deinstallation als auch die Installation des Adapters als auch Instanzen verläuft immer fehlerfrei. Angeblich werden auch Objekte angelgt bzw wieder gelöscht bei der Installation. Trotzdem habe ich unter oBjekte immer nur einen leeren Zendure-Solarflow Ordner.
Verbinden tut die Instanz sich auch einwandfrei und wird grün.

Andere Adapter legen noch einwandfrei Instanzen an und erstellen Unterodner entsprechend.

Jemand eine Idee woran das liegen könnte und was ich noch tun kann?

raspberrypi 4
js Controller 5.0.17
arm64
RAM 7.63 GB
Node.js v18.19.0
NPM 10.2.3

Wunni73, 2024-01-27:
Ich habe die gleichen Probleme. Habe bei der Installation auch das Problem das der IOBroker sich aufhängt. 
Peter hat mir erklärt das wir keine Daten erhalten weil wir auf den EU Server sind, und das IOBroker Tool für den Global Server ist.

PS: Ich schreibe es auf den EU Server um.  Berichte dir wenn es geklappt hat.

nograx, 2024-03-14:
Die neuste Version unterstützt nun auch den EU server. Kann in den Adaptereinstellungen umgestellt werden. Gerne mal testen.

nograx, 2024-03-15:
Version 1.2.2 im Beta Kanal unterstützt den EU server und funktioniert laut mehreren Berichten 1a. Issue wird geschlossen.

closed 2024-03-15 by nograx