nograx/ioBroker.zendure-solarflow #24: Error „Retrieving device failed“ ever now and then

andiweli, 2024-02-16:
**Describe the bug**  
Error in Log „ zendure-solarflow.0
2024-02-16 19:53:55.324	error	Retrieving device failed!“

I am logged in and everything seems to work.
But every now and then I see this error in the Log…

andiweli, 2024-02-19:
"Every now and then" seems to be more often now...
[image]

nograx, 2024-02-28:
I have adjusted the timeout values to give the server more time to respond. Please check version 1.1.4 if the behavior is better now.

andiweli, 2024-02-28:
So far no errors anymore :) Thanks

nograx, 2024-02-29:
Perfect, will close this for now.

closed 2024-02-29 by nograx