nograx/ioBroker.zendure-solarflow #26: Auswahl von Europa Server

manigl, 2024-02-27:
Könnten Sie den Adapter für den Europa Server anpassen?

nograx, 2024-03-06:
Hallo, 

mein Problem dabei ist das mein eigenes System auf dem globalen Server registriert ist. Um den Adapter anzupassen müsste ich das System auf den Europa Server umstellen. Das ist a) aufwendig und b) kann ich das System in der Zeit nicht über meinen ioBroker steuern. Daher kommt dieser Schritt für mich aktuell nicht in Frage. Tut mir leid. 

Sollte ich irgendwann günstig einen zweiten Hub in die Finger bekommen könnte ich das Thema noch mal angehen.

Gruß,
Peter

nograx, 2024-03-07:
Ich werde im Laufe des Abends eine Version bereitstellen mit welchem der Europa Server getestet werden kann. Bin hier dann auf Feedback angewiesen.

manigl, 2024-03-07:
Perfekt, werde Feedbacks übermitteln. Danke

nograx, 2024-03-11:
Hi, theoretisch könnte der EU server jetzt genutzt werden. Login war bei mir zumindest möglich (wenn ich da keinen Bug eingebaut habe). Kann also gerne mal getestet werden.

nograx, 2024-03-12:
Das ist interessant, Login scheint ja funktioniert zu haben. Die Fehlermeldung deutet darauf hin das mit der Geräte ID was nicht passt. 

Wäre es möglich das du das Gerät mal für meinen Account freigeben könntest?

nograx, 2024-03-12:
> Moin, habe mir jetzt extra um eine erste Rückmeldung geben zu können mal ein Accound Gemacht. Die umstellung ging bei mir fast Problem los. Das einzige was Probleme gemacht hat war nicht die Serverauswahl sondern das unter Calculation (ordner in iobroker) keine werte angelegt werden und der dann im Protokoll dieses vermerkt. Diese habe ich dann manuel angelegt. Ansonsten läuft. 👍

Dh der Adapter läuft bei dir über den EU Server?

funflo, 2024-03-12:
> Dh der Adapter läuft bei dir über den EU Server?

Jo, hatte ich heute morgen direkt umgestellt von global auf EU, da ich eh noch am rumbasteln war.
ich hatte hierzu aber den adapter (auch die bereits angelegten werte) gelöscht und den neu eingerichtet.

manigl, 2024-03-12:
> Das ist interessant, Login scheint ja funktioniert zu haben. Die Fehlermeldung deutet darauf hin das mit der Geräte ID was nicht passt.
> 
> Wäre es möglich das du das Gerät mal für meinen Account freigeben könntest?

Welche Mail Adresse soll ich hinterlegen?

nograx, 2024-03-13:
> > Das ist interessant, Login scheint ja funktioniert zu haben. Die Fehlermeldung deutet darauf hin das mit der Geräte ID was nicht passt.
> > Wäre es möglich das du das Gerät mal für meinen Account freigeben könntest?
> 
> Welche Mail Adresse soll ich hinterlegen?

Kannst du mich per E-Mail an Peter.frommert@outlook.com kontaktieren? Da kann ich dir dann die für den passenden Account nennen.

funflo, 2024-03-13:
in der 1.1.23 hatte ich keine probleme in der Aktuellen (1.2.1) hatte ich keine Daten mehr bekommen.
ich bin erstmal wieder zur 1.1.23 zurück gegangen.
hier mal einen auszug aus dem log:

info	[onConnected] Connected with MQTT!
debug	[connectMqttClient] Found 0 SolarFlow devices.
debug	[connectMqttClient] Connecting to MQTT client...
info	[onReady] Found 0 SolarFlow devices.
debug	[getDeviceList] Getting device list from Zendure Rest API!
info	[login] Login to Zendure Rest API successful!
debug	[onReady] Using server eu

nograx, 2024-03-13:
> in der 1.1.23 hatte ich keine probleme in der Aktuellen (1.2.1) hatte ich keine Daten mehr bekommen. ich bin erstmal wieder zur 1.1.23 zurück gegangen. hier mal einen auszug aus dem log:
>
> info [...]

Hast du zufällig einen Hub 2000?

funflo, 2024-03-13:
Ne, hub1200 (den „alten“)

nograx, 2024-03-14:
Wäre es für dich ok mir beide Hubs mal freizugeben? Falls ja kontaktier mich doch bitte unter Peter.frommert@outlook.com, dann gebe ich dir die Daten von meinem dev Account… das Problem von manigl wurde so gelöst.

nograx, 2024-03-15:
@funflo ich nehme an bei dir funktioniert es jetzt mit Version 1.2.2?

funflo, 2024-03-15:
Jo, hatte eben das update über github gemacht läuft 👍 
mir wurde das nicht zur installation so vorher angeboten weswegen ich es erst jetzt gemacht hatte.