nograx/ioBroker.zendure-solarflow #27: PV Input 1 and 2 not synchronized anymore

andiweli, 2024-03-01:
**Describe the bug**  
PV Input 1 and 2 aren't shown anymore.

[image]

Don't know if it has to do with yesterdays new Hub Firmware or yesterdays Adapter upgrade to 1.1.4.

nograx, 2024-03-01:
Seems the issue is located at the new firmware from zendure. Will investigate this weekend.

andiweli, 2024-03-01:
> Seems the issue is located at the new firmware from zendure. Will investigate this weekend.

Seems they renamed the parameters "solarPower1" and "solarPower2" now.

nograx, 2024-03-01:
I've compiled version 1.1.9 which will be available in the stable channel this night. Or maybe you can install it via the "Install from Github" option in iobroker. The change with the solarPower1 and 2 is untested from my side. Maybe you can verify the change? I'm not able to install the firmware update before sunday.

andiweli, 2024-03-01:
[image]

Seems to work :) Thanks for the fast fix.

nograx, 2024-03-01:
Seems to be very dark on your location. Will close this issue - thanks for testing.

closed 2024-03-01 by nograx

andiweli, 2024-03-01:
Raining all the day