nograx/ioBroker.zendure-solarflow #28: Error Retrieving device

andiweli, 2024-03-01:
**Describe the bug**  
Seems the issue from last week is back with the latest Firmware upgrade... 🤷🏻‍♂️

zendure-solarflow.0
2024-03-01 19:31:15.529	error	Retrieving device failed!

But seems to work so far even with this issue.

andiweli, 2024-03-03:
Got some more errors today... 

`zendure-solarflow.0 | 2024-03-03 18:53:00.593 | error | DB closed
-- | -- | -- | --
zendure-solarflow.0 | 2024-03-03 18:53:00.592 | error | Error: DB closed at Redis.sendCommand (/opt/iobroker/node_modules/ioredis/built/redis/index.js:636:24) at Redis.get (/opt/iobroker/node_modules/ioredis/built/commander.js:122:25) at ObjectsInRedisClient._getObject (/opt/iobroker/node_modules/@iobroker/db-objects-redis/build/lib/objects/objectsInRedisClient.js:2413:37) at Immediate.<anonymous> (/opt/iobroker/node_modules/@iobroker/db-objects-redis/build/lib/objects/objectsInRedisClient.js:2460:33) at process.processImmediate (node:internal/timers:478:21)
zendure-solarflow.0 | 2024-03-03 18:53:00.589 | error | unhandled promise rejection: DB closed
zendure-solarflow.0 | 2024-03-03 18:53:00.586 | error | Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
zendure-solarflow.0 | 2024-03-03 18:53:00.580 | error | DB closed
zendure-solarflow.0 | 2024-03-03 18:53:00.579 | error | Error: DB closed at Redis.sendCommand (/opt/iobroker/node_modules/ioredis/built/redis/index.js:636:24) at Redis.get (/opt/iobroker/node_modules/ioredis/built/commander.js:122:25) at StateRedisClient.setState (/opt/iobroker/node_modules/@iobroker/db-states-redis/build/lib/states/statesInRedisClient.js:521:40) at ZendureSolarflow._setState (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:5537:76)
zendure-solarflow.0 | 2024-03-03 18:53:00.578 | error | unhandled promise rejection: DB closed
zendure-solarflow.0 | 2024-03-03 18:53:00.577 | error | Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
zendure-solarflow.0 | 2024-03-03 18:53:00.574 | error | DB closed
zendure-solarflow.0 | 2024-03-03 18:53:00.573 | error | Error: DB closed at Redis.sendCommand (/opt/iobroker/node_modules/ioredis/built/redis/index.js:636:24) at Redis.get (/opt/iobroker/node_modules/ioredis/built/commander.js:122:25) at StateRedisClient.setState (/opt/iobroker/node_modules/@iobroker/db-states-redis/build/lib/states/statesInRedisClient.js:521:40) at ZendureSolarflow._setState (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:5537:76) at runNextTicks (node:internal/process/task_queues:60:5) at process.processImmediate (node:internal/timers:447:9)
zendure-solarflow.0 | 2024-03-03 18:53:00.566 | error | unhandled promise rejection: DB closed
zendure-solarflow.0 | 2024-03-03 18:53:00.559 | error | Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().`

andiweli, 2024-03-05:
Seems this issue is also existing in 1.1.14...

```

	Zeit

	
Nachricht

zendure-solarflow.0
2024-03-05 18:44:00.577	error	DB closed

zendure-solarflow.0
2024-03-05 18:44:00.576	error	Error: DB closed at Redis.sendCommand (/opt/iobroker/node_modules/ioredis/built/redis/index.js:636:24) at Redis.get (/opt/iobroker/node_modules/ioredis/built/commander.js:122:25) at StateRedisClient.setState (/opt/iobroker/node_modules/@iobroker/db-states-redis/build/lib/states/statesInRedisClient.js:521:40) at ZendureSolarflow._setState (/opt/iobroker/node_modules/@iobroker/js-controller-adapter/build/lib/adapter/adapter.js:5537:76)

zendure-solarflow.0
2024-03-05 18:44:00.574	error	unhandled promise rejection: DB closed

zendure-solarflow.0
2024-03-05 18:44:00.571	error	Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().

```

nograx, 2024-03-06:
Releasing a new version this moment. I think I fixed this...

andiweli, 2024-03-07:
And a new one...

zendure-solarflow.0 | 2024-03-07 18:55:16.594 | error | [onReady] Logon error at Zendure cloud service! Error: [login] Failed to login to Zendure REST API!
-- | -- | -- | --
zendure-solarflow.0 | 2024-03-07 18:55:16.590 | error | AxiosError: timeout of 10000ms exceeded

andiweli, 2024-03-08:
@nograx 1.1.18 nearly instantly goes red in adapters here...

zendure-solarflow.0 | 2024-03-08 10:24:07.496 | error | [onReady] Logon error at Zendure cloud service! Error: [login] Failed to login to Zendure REST API!
-- | -- | -- | --
zendure-solarflow.0 | 2024-03-08 10:24:07.491 | error | AxiosError: Request failed with status code 400

Webranger2, 2024-03-08:
> @nograx 1.1.18 nearly instantly goes red in adapters here... A few seconds green, then turns to red with below mentioned error.
> 
> zendure-solarflow.0	2024-03-08 10:24:07.496	error	[onReady] Logon error at Zendure cloud service! Error: [login] Failed to login to Zendure REST API!
> zendure-solarflow.0	2024-03-08 10:24:07.491	error	AxiosError: Request failed with status code 400

Login Daten komplett neu eingeben hat bei mir geholfen.

nograx, 2024-03-08:
Seid aber schnell beim Updaten... eventl mal mit der 1.1.20 versuchen. Oder Daten neu eingeben...

andiweli, 2024-03-08:
Jap kann ich glaub ich bestätigen, Kennwort neu eingegeben und aktuell kein Error 400 mehr, mit 1.1.20 getestet.

andiweli, 2024-03-08:
DB Closed ist mit 1.1.20 noch vorhanden seh ich grad.

andiweli, 2024-03-11:
Error retrieving device seems gone now.

closed 2024-03-11 by andiweli