nograx/ioBroker.zendure-solarflow #34: [onReady] Logon Error

andiweli, 2024-03-11:
**Describe the bug**  
[onReady] Logon Error at Zendure cloud service

**To Reproduce**  
Appears randomly... in that case 6 minutes after the "DB closed" error.
Sometimes happens "alone".

**Screenshots & Logfiles**  

[image]

**Versions:**  
 - Adapter version: 1.1.22
 - JS-Controller version: 5.0.19
 - Node version: 18.19.1
 - Operating system: Raspberry OS Lite Bookworm

nograx, 2024-03-11:
I think the DB closed error ist the same as this logon error, resulting of a timeout of the web request. Normally this can be ignored because the web api is just used for login and retrieving the device list. The "live" data comes from the Mqtt service.

funflo, 2024-03-16:
[onReady] Logon error at Zendure cloud service! Error: [login] Failed to login to Zendure REST API!
error	AxiosError: Request failed with status code 500

Da mein Englisch eher nicht vorhanden ist, bitte ich um entschuldigung das ich auf Deutsch schreibe.
Ich hatte heute aufgrund von wartungsbedingtem Neustart diese Fehler im log (heute um ca. 12:10). Danach hatte ich keine daten mehr bekommen und eine Automatik konnte auch nix mehr stellen. Diese beiden fehler kann ich reproduzieren in dem ich den adapter mehrmals (mit glück aber manchmal schon beim ersten mal) neustart.

nograx, 2024-03-16:
Auf dem global Server gab es heute mir exakt den selben Fehler, allerdings morgens gegen 10 Uhr. Ich nehme an das Zendure da an den Servern gearbeitet hat… angekündigt wurde die Serverlandschaft zu vergrößern. Bitte weiter beobachten.

andiweli, 2024-08-07:
Never happened again since weeks.

closed 2024-08-07 by andiweli