nograx/ioBroker.zendure-solarflow #35: energyWh shows wrong Watt

andiweli, 2024-03-13:
**Describe the bug**  
energyWh (Energie in den Batterien (Wh)) shows 7342 Wh.
I only have a max. of 4800 Wh (3x AB1000 + 1x AB2000).

**Versions:**  
 - Adapter version: 1.1.22
 - JS-Controller version: 5.0.19
 - Node version: 18.19.1
 - Operating system: Raspberry OS Lite Bookworm

nograx, 2024-03-13:
Please install version 1.2.1 via Github and let the hub charge to 100%. This will set the current energy in batteries as the max value.

andiweli, 2024-03-14:
@nograx shows 10066,93087389 now. More than 2 times my maximum Wh 😳

nograx, 2024-03-14:
There is a new version online where you can modify that state. Currently I don't know why this happens. The value if working fine for my system...

nograx, 2024-04-02:
Hi @andiweli, I have released a prerelease version to GitHub and npm. Could you try these version and bring the HUB to the discharge limit? Then it should reset the maximum Wh values and calculate the correct values afterwards...

andiweli, 2024-04-02:
HelloThanks for the update - I will try as soon as possible and we have more sun - perhaps the next weekend. Thanks

nograx, 2024-04-12:
Can I close this issue or does it still persist?

closed 2024-04-12 by andiweli