nograx/ioBroker.zendure-solarflow #40: Please fix issues reported by adapter checker

mcm1957, 2024-04-01:
👀 [W171] "common.title" is deprecated in [io-package.json](https://github.com/nograx/ioBroker.zendure-solarflow/blob/master/io-package.json). Please remove from [io-package.json](https://github.com/nograx/ioBroker.zendure-solarflow/blob/master/io-package.json).
 👀 [W156] Adapter should support admin 5 UI (jsonConfig) if you do not use a React based UI
 👀 [W181] "common.license" in [io-package.json](https://github.com/nograx/ioBroker.zendure-solarflow/blob/master/io-package.json) is deprecated. Please define object "common.licenseInformation"
 👀 [W173] "@iobroker/adapter-core" (^3.0.4) should be 3.0.6 or newer - please update
 👀 [W115] common.tier is required in [io-package.json](https://github.com/nograx/ioBroker.zendure-solarflow/blob/master/io-package.json). Please check https://github.com/ioBroker/ioBroker.docs/blob/master/docs/en/dev/objectsschema.md#adapter.

nograx, 2024-04-26:
Fixed all poins except W156 ... I'm using React based UI.

closed 2024-04-26 by nograx

mcm1957, 2024-04-26:
Thanks. 
Everythings fine.