nograx/ioBroker.zendure-solarflow #41: keine aktuellen Daten nach RefreshTokenTimerJob

HC-Hofer, 2024-04-11:
Hallo.

Hatte heute wieder das Problem das ich keine aktuellen Werte mehr bekomme.
Im Log steht zwar das er verbunden ist aber bekomme keine Daten.
Nach neustart des Adapters funktioniert alles wieder.

[image]

14:44 Uhr : RefreshToken
15:48 Uhr : Zendure auf Bypass
17:30 -17:45 Uhr : Zendure Bypass aus
**17:44 Uhr** : RefreshToken

Mein Bypass Modus steht auf Automatik.

Hoffe diese infos können weiterhelfen.

Lg HC-Hofer

nograx, 2024-04-12:
Dh. du hast direkt nach 17:44 keine Daten mehr bekommen? Oder wo ist die Datenlücke?

HC-Hofer, 2024-04-12:
Ich habe nach 17:44 keine Daten mehr bekommen. Timestamp von der Daten Aktualisierung hat sich auch nicht mehr geändert.

nograx, 2024-04-18:
Hi @HC-Hofer,

ich konnte bei mir gestern das selbe beobachten, ich habe 2 Hubs im Einsatz und bei genau einem habe ich direkt nach dem RefreshToken keine Daten mehr bekommen - beim anderen trudelten munter Daten rein. Neustart vom Adapter und es lief sofort wieder... habe da nur noch keine Erklärung für...

Werde aber einen Workaround schaffen - wenn Gerät online und keine Daten für 5 Minuten - dann neu verbinden...

Gruß,
Peter

HC-Hofer, 2024-04-18:
Ich danke dir für deine Bemühungen

Lg,
Christopher

HC-Hofer, 2024-05-24:
Hallo Peter.
Hatte gestern wieder das gleiche Problem. Könnte aber auch sein das es ein Server Problem bei Zendure ist (wie ich auf der Facebook Seite gelesen habe), da andere auch mit den Server Problemen zu tun haben.

nograx, 2024-07-31:
Hi, besteht das Problem bei dir noch?

closed 2024-09-06 by nograx