nograx/ioBroker.zendure-solarflow #45: Heutige Ladung / Einspeisung zur / von Batterie summiert über Tage

andiweli, 2024-05-21:
**Describe the bug**  
Die 6 Werte für:
outputHomeEnergyToday (Wh und kWh)
outputPackEnergyToday (Wh und kWh)
packInputEnergyToday (Wh und kWh)
...resetten nicht um Mitternacht, sie summieren sich bei mir.

**Expected behavior**  
Um 23:59:59 sollten die Werte auf 0 gesetzt werden.

**Screenshots & Logfiles**  
[image]
Meine 4 Panele täglich vom untersten Limit von 10% soweit wie möglich rauf bis 100%.
Es sind 4 Akkus, in Summe 4.800 kWh (AB1000 und AB2000 gemischt).

**Versions:**  
 - Adapter version: 1.6.1
 - JS-Controller version: 5.0.19
 - Node version: 18.20.2
 - Operating system: RaspiOS Lite

andiweli, 2024-06-17:
Perhaps it would also help, to make the values writeable.
So the user can start a script at 23:59:59 so set the values to zero...?

andiweli, 2024-07-25:
Seems to work now since a few days.

andiweli, 2024-07-29:
Since two days the issue is back again. 
Today values are summed up from today and yesterday.

nograx, 2024-07-31:
Could you check if the adapter was (re)started around 0:00h?

andiweli, 2024-07-31:
I was on v1.7.2, currently updating to v1.7.4.

I've set the adapter to manually restart every 4 hours (starting at 0:00) because of the sporadic disconnects due to https://github.com/nograx/ioBroker.zendure-solarflow/issues/50

So currently I have only the manual restarts (which I have deleted now).

What will be the effect of the restart around 0:00?

nograx, 2024-07-31:
The values will be reset at 0:00, and when you manually restart the adapter this trigger will not work (as the adapter is not running at that specific time). The adapter will restart every 2 hours by itself (instead of just reconnect to Zendure), so I think it is not necessary to do this anymore...

andiweli, 2024-07-31:
Thanks for the clarification.
I am on 1.7.4 now and have deleted my manually restarts of the adapter.

I will check it tomorrow.
Currently I have 20,81 kWh charging (from over 5 days) and 19,09 kWh output (from 5 days).
So I assume tomorrow it should be correct again :)

Thanks!

andiweli, 2024-08-01:
@nograx has been reset to 0 at 0:00 👍🏻

closed 2024-08-07 by andiweli