nograx/ioBroker.zendure-solarflow #46: EnergyWhMax can't be overwritten, shows wrong value at 100%

andiweli, 2024-05-23:
**Describe the bug**  
4.800 Wh in sum at 100% but EnergyWhMax always resets to some value around 3xxx Wh when charging is reached at 100%.

Btw - what's the difference to EnergyWh?

**Expected behavior**  
Set a fixed value.

**Versions:**  
Adapter version: 1.6.1
JS-Controller version: 5.0.19
Node version: 18.20.2
Operating system: RaspiOS Lite

nograx, 2024-08-06:
With version 1.8.1 the maximum of battery capacity is used as the max value. Please try if issue is fixed with this version.

andiweli, 2024-08-06:
How does the adapter know, what's the max capacity?
I installed 1.8.1 and currently it shows 3132,xxx Wh "energyWhMax" and 2990,xxx "energyWh".

Batteries are 4 (AB1000/2000 mixed) and should be theoretically max. 4800 Wh.
Currently it is at 95% discharging state... so only 5 down from 100.

nograx, 2024-08-06:
There was is little bug in 1.8.1 (was working in dev environment - but not in production), fixed it with 1.8.2 and now it seems to work. It will check the connected batteries at adapter start (with the serial id we can check if it's a AB1000 or AB2000) and this is the maximum value. Maybe over time the max value can be smaller - but not bigger.

andiweli, 2024-08-06:
Hmm, energyWhMax still shows 3132,xxx 
SoC shows 13,5% but is 34% in reality

nograx, 2024-08-07:
The calculation needs at least one full charge and then down to the discharge limit or if the battery protect is activated to the battery voltage of 46,1V.

andiweli, 2024-08-07:
I see, thanks for the clarification!
Will keep an eye on it :)

andiweli, 2024-08-07:
I don't know if that has something to do with the latest update 1.8.2
But Log shows very strange errors since today...

zendure-solarflow.0 | 2024-08-07 15:52:50.116 | error | [JSON PARSE ERROR] ~))��)�_HARDWARE_FLOW_CONTROL(USART_InitStruct->HardwareFlowControl)..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_usart.c�� <����< �C��$6n#E�H��L?������-��>��
-- | -- | -- | --
zendure-solarflow.0 | 2024-08-07 15:52:49.890 | error | [JSON PARSE ERROR] ~&))�nstech\firmware\n32g43x_std_periph_driver\src\n32g43x_exti.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_flash.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_gpio.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_iwdg.cIS_LPUART_HARDWARE_FLOW_CONTROL(LPUART_InitStruct->HardwareFlowControl)..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_lpuart.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_pwr.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_rcc.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_rtc.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_spi.cIS_USARTfz
zendure-solarflow.0 | 2024-08-07 15:52:49.657 | error | [JSON PARSE ERROR] ~#&)��lA��A�A��@�!�@���@��@O�@���@���@��@+��@�8�@R'�@vO�@O��@9E�@e�@��@aT�@��@9E�@��@�~@?v@<�n@؁g@Nb`@��Y@H�R@�zL@�MF@d]@@b�:@+5@w�/@F�*@S�%@�� @�@$�@d;@�@�"@#J@��@���?��?C��?H��?HP�?���?�;�?m��? ��?���?3Ĺ?�H�?�?��?ё�?ŏ�?/�?\|�?[Ӝ?�Ԙ?��?4�?"��?u�?F��?=�?� �?q=z?�`t?ın?0*i?�c??5^?��X?��S?2�N?�J?xzE?\|�@?ё<?^K8?��3?�/?q�+?�'?}�#? IS_NVIC_SUB_PRIORITY(NVIC_InitStruct->NVIC_IRQChannelSubPriority)IS_NVIC_PREEMPTION_PRIORITY(NVIC_InitStruct->NVIC_IRQChannelPreemptionPriority)..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\misc.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_adc.c..\lib\lib_nationstech\firmware\n32g43x_std_periph_driver\src\n32g43x_can.c..\lib\lib_natio'�
zendure-solarflow.0 | 2024-08-07 15:52:49.418 | error | [JSON PARSE ERROR] ~ #)��d�d?d<d��d��d5d2ddd<d7d��d��dZdPdAd<deddddd^d,ddddd��d����d�,ddd�d�d�dB� d���dAd<d��d��d7d2dddFdAd��d��dsdUdFdAdfddddd�d ddddd�������d��dd:d�dwd�d�d d� d�dFd<d��d��dAd2d��ddKdFd��d��dxdUdKdFdgddddd�d�dddddp�����^d��dd��-A-722<<2(2<PPdddPP2<PPPdddP<22PPdPPdPPd0�B�}�B�u�B�B<�B���B�<�B��B�0�BёB�֊B�By)B��tBcnjB[�`B!�TB�IB�m?B��5B��,B��$B �B�B�B� B�VBC��A�5�A5o�A�>�A���A�-�A0̿A�ѷA�8�A�-�A�p�A]��A&ӕAQ�A��A$(�AL7xA�mA��bAI.\AGrUA�NAy�GA&AAT�9AZ�2AyX,A1&A A�<AǺA`vAa�

nograx, 2024-08-07:
Yeah These are messages generated when a Firmware Update is in Progress. You can ignore them.

andiweli, 2024-08-07:
> The calculation needs at least one full charge and then down to the discharge limit or if the battery protect is activated to the battery voltage of 46,1V. Then the value will show your real useable capacity.

I understand discharge limit - thats the 10% I set. 
But what‘s the „battery protect“?

nograx, 2024-08-08:
"Battery protect" is the "low voltage blockt" from settings. It will set the outputLimit to 0 if a voltage of 46,1V is reached (works only if controlled by adapter). Sometimes Zendure thinks the battery has more energy and if you discharge with a high output it's possible you brick your battery (deep discharge).

nograx, 2024-09-06:
I think this is fixed now.

closed 2024-09-06 by nograx