nograx/ioBroker.zendure-solarflow #52: Hyper and ACE units don't seem to appear.

ppitkin, 2024-07-12:
Firstly - A BIG thankyou for developing this interface. definitely more reliable than using the MQTT feed.

**Describe the bug**  
The new Hyper and ACE devices do not seem to be picked up and delivered in the feed.

Zendure has just started sending the information via the MQTT connection - so I am assuming it should also be available via the API.
I do not however see the devices appearing in the IOBroker Object list.

I do receive the data when using the MQTT connection.

Any chance that you could take a look at the issue.

If you don't have those devices I would be more than happy to assist with testing
I have:
Solarflow 1200 (3 units)
Solarflow 2000 (3 units)
AIO (1 unit)
Hyper 2000 (3 units)
ACE (1 unit)

nograx, 2024-07-26:
There will be an update in the next days which includes support for the ACE 1500 and the Hyper 2000.

nograx, 2024-07-27:
Update 1.7.0 is available. Please Test this version, as I don‘t have a ace of hyper for direct testing.

ppitkin, 2024-07-29:
version 1.7.0 doesn't appear to be showing up in the list of available adaptors - its still the old version. I also tried switching to the beta repository and it still didn't appear :-(

nograx, 2024-07-29:
Hm, that's strange. Can you try to use the "Install from Github" option in ioBroker (Advanced Mode) and install the current version from Github? 

The Beta repo should contain version 1.7.2 from tomorrow.

ppitkin, 2024-07-31:
Sorry but i have no idea how to install something in iobroker directly from github. never done that before.

What I did was select the adaptor an tell it to install a specific version and then i specified 1.7.2
It appeared to work and the adaptor is now saying its version 1.7.2

However, I still see that its detecting no additional devices - It says in the log that it detected 7 solarflow devices (that would be my 3 1200 Hubs, my 3 2000 hubs and my AIO.  It is not seeing the  3 Hypers - although I am seeing those devices on the direct MQTT feed that I have running in parallel to your plugin.

is it possible to but the adaptor into a debug logging mode so that we can determine why it doesn't see the devices?

Also for the existing Solarflows it has a new "control" value (control.setInputLimit) - but it doesn't create the object and therefore throws an error in the log.

<html>
<body>

zendure-solarflow.0 | 2024-07-31 12:02:00.286 | warn | State "zendure-solarflow.0.A8yh63.YrQ1X08v.control.setInputLimit" has no existing object, this might lead to an error in future versions
-- | -- | -- | --
zendure-solarflow.0 | 2024-07-31 12:01:59.269 | warn | State "zendure-solarflow.0.A8yh63.14CWUD28.control.setInputLimit" has no existing object, this might lead to an error in future versions
zendure-solarflow.0 | 2024-07-31 12:01:57.725 | warn | State "zendure-solarflow.0.A8yh63.Yy912A87.control.setInputLimit" has no existing object, this might lead to an error in future versions
zendure-solarflow.0 | 2024-07-31 12:01:57.113 | warn | State "zendure-solarflow.0.73bkTV.hCNDkQfy.control.setInputLimit" has no existing object, this might lead to an error in future versions
zendure-solarflow.0 | 2024-07-31 12:01:56.253 | warn | State "zendure-solarflow.0.73bkTV.Mu49k0SQ.control.setInputLimit" has no existing object, this might lead to an error in future versions
zendure-solarflow.0 | 2024-07-31 12:01:54.773 | warn | State "zendure-solarflow.0.73bkTV.M5qsmu87.control.setInputLimit" has no existing object, this might lead to an error in future versions
zendure-solarflow.0 | 2024-07-31 12:01:52.357 | warn | State "zendure-solarflow.0.yWF7hV.NL9W6xs0.control.setInputLimit" has no existing object, this might lead to an error in future versions
zendure-solarflow.0 | 2024-07-31 12:01:47.374 | info | [onConnected] Connected with MQTT!
zendure-solarflow.0 | 2024-07-31 12:01:46.478 | info | [onReady] Found 7 SolarFlow device(s).
zendure-solarflow.0 | 2024-07-31 12:01:45.892 | info | [login] Login to Zendure Rest API successful!
zendure-solarflow.0 | 2024-07-31 12:01:43.818 | info | starting. Version 1.7.2 in /opt/iobroker/node_modules/iobroker.zendure-solarflow, node: v18.20.3, js-controller: 5.0.19

</body>
</html>

nograx, 2024-07-31:
It's possible by activating the "advanced mode" on the command bar when you navigated to "adapters". Then there will be an Icon to install from Github (just use the dropdown to select the Zendure Solarflow adapter). 

Otherwise, version 1.7.3 should be in the beta repository... 1.7.4 from tomorrow... this will also fix the "setInputLimit" issue.

Do you use the adapter in "Fallback"-Mode? I'm not sure if this will work with the Hyper.

ppitkin, 2024-07-31:
Many thanks. Found it. Never used that option before.
It installed 1.7.4

still no new object created in the objects view. Still says only found 7 devices.

No, I don't have the "fallback mode" selected.

<html>
<body>

zendure-solarflow.0 | 2024-07-31 14:45:51.655 | info | [onConnected] Connected with MQTT!
-- | -- | -- | --
zendure-solarflow.0 | 2024-07-31 14:45:50.740 | info | [onReady] Found 7 SolarFlow device(s).
zendure-solarflow.0 | 2024-07-31 14:45:50.010 | info | [login] Login to Zendure Rest API successful!
zendure-solarflow.0 | 2024-07-31 14:45:47.026 | info | starting. Version 1.7.4 (non-npm: nograx/ioBroker.zendure-solarflow) in /opt/iobroker/node_modules/iobroker.zendure-solarflow, node: v18.20.3, js-controller: 5.0.19
zendure-solarflow.0 | 2024-07-31 14:45:43.250 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
zendure-solarflow.0 | 2024-07-31 14:45:43.249 | info | terminating
zendure-solarflow.0 | 2024-07-31 14:45:43.248 | info | Got terminate signal TERMINATE_YOURSELF

</body>
</html>

ppitkin, 2024-07-31:
[image]

ppitkin, 2024-07-31:
If I "Shared" my Hypers with you in the Zendure app would you then be able to pick them up in your feed?

nograx, 2024-08-01:
I have a suspicion: Do you use your 3 Hypers as a cluster? Maybe the cluster will appear as a different product... 

Are you a member in the iobroker Forum? Maybe you can send me a PN there and I will give you the e-mail for my Zendure dev account.

ppitkin, 2024-08-01:
no I'm not clustered - each hyper is on its own phase.
I will try to reach out to you in the forum

ppitkin, 2024-08-02:
Issues now solved in latest release - Thanks. You can close the ticket.

closed 2024-08-02 by ppitkin