nograx/ioBroker.zendure-solarflow #61: Login Errors since a few days

andiweli, 2024-08-31:
Got this error again since a few days:

2024-08-31 11:12:42.867 - error: zendure-solarflow.0 (111473) AxiosError: timeout of 10000ms exceeded

2024-08-31 11:12:42.871 - error: zendure-solarflow.0 (111473) [onReady] Logon error at Zendure cloud service! Error: [login] Failed to login to Zendure REST API!

I am using 1.8.5 adapter version. 
rest of ioBroker is up to date.

nograx, 2024-08-31:
I have the same Problem singe 2 days. So this is a Zendure Problem…

nograx, 2024-09-06:
I've released version 1.8.6 which has a new function which will retry the login process if it fails. Maybe this will solve this problem, as it's hard to test. Please have a look.

nograx, 2024-09-26:
Did you test the latest beta version? It seems that the login behavior is much better now...

andiweli, 2024-09-26:
1.8.8 and no login errors anymore 👍🏻

closed 2024-09-26 by nograx