nograx/ioBroker.zendure-solarflow #65: Exposing objects does not properly work for control section/objects

FlySmack, 2024-09-19:
**Describe the bug**  
I try to expose the objects of zendure-solarflow with the help of MQTT for further automation. The iobroker implementation of the Zendure API is the most stable and complete one I could find.

The zendure-solarflow adapter works like it should be. I have installed the MQTT server/client adapter in iobroker and configured it the way, so that the zendure-solarflow objects are exposed by MQTT.

I use Home Assistant to subscribe to the corresponding MQTT topic of zendure-solarflow from the MQTT server running in iobroker and also this works perfectly with receiving the data.

As soon as I try to publish a change of the control objects through MQTT, the change is just partially handled.
Example:
Publishing a change for zendure-solarflow.0.XXXX.YYYY.control.setOutputLimit is reflected by the zendure-solarflow adapter, but the corresponding object zendure-solarflow.0.XXXX.YYYY.outputLimit will not be set, like it is published to the Zendure backend.

If I set a new value through iobroker UI, it works as expected.

Any hint, or idea what might be the cause of this behaviour would help a lot. I do not rule out, that I have a configuration mistake somewhere.

**To Reproduce**  
Steps to reproduce the behavior:
1. Installing the zendure-solarflow adapter and configure it
2. Install the MQTT server adapter and publish the state of zendure-solarflow.0.*
3. Using a MQTT client, connect to the MQTT server and subscribe to the following topics:
```
topic zendure-solarflow/# in 0
topic zendure-solarflow/0/XXXX/YYYY/control/# both 0
```
4. From the MQTT client, publish a new value to MQTT topic zendure-solarflow/0/XXXX/YYYY/control/setOutputLimit
5. Change is visible in the zendure-solarflow object zendure-solarflow.0.XXXX.YYYY.control.setOutputLimit but zendure-solarflow.0.XXXX.YYYY.outputLimit does not reflect the value, like the change is not passed to the Zendure backend. The Hyper 2000 (this is what I use) also does not react on the change (checked with the Zendure App).

**Expected behavior**  
I would expect that solarflow.0.XXXX.YYYY.outputLimit reflects the change and the Zendure device reacts accordingly.

**Screenshots & Logfiles**  
MQTT Server configuration
[image]
[image]

State before MQTT publish
[image]
[image]

State after MQTT publish
[image]
[image]

Logfiles do not provide any useful information. Tried it with setting the adapter to "silly" and not even the change of setOutputLimit is visible in the logs.

**Versions:**  
 - Adapter version: Zendure Solarflow: 1.8.5 MQTT Server: 5.2.0
 - JS-Controller version: Not sure where to find this in the docker version of iobroker
 - Node version: v20.17.0
 - Operating system: docker (official image - v10.0.0)

**Additional context**  
Add any other context about the problem here.

nograx, 2024-09-19:
Hi, 

maybe the problem here is that the state change in ioBroker is sent with the ack flag, as I think these changes will not trigger the change in the ioBroker adapter. The ack flag will be set by the adapter when the change is carried out.

Is it possible to make the change of the ioBroker state without the ack flag?

Kind regards,
Peter

FlySmack, 2024-09-19:
Hi,

Do you refer to the MQTT server setting "Send states (ack=true) too" from my screenshot, or do you refer to something else? I'm not familiar with iobroker and just started to use it a couple of days ago when I discovered your Zendure implementation.

Kind regards,
Torsten

FlySmack, 2024-09-19:
I just disabled the function in the MQTT server and tried it again with the same result. Without the "Send states (ack=true) too" setting turned on, the MQTT server just send 4-5 updates and then you don't receive any further update from the zendure-solarflow adapter through the MQTT server. I already discovered this behaviour when I initially started to perform tests with the MQTT server. That's the reason, why I had turned it on because then I get constantly updates through MQTT when any value of the zendure-solarflow objects are changing.

FlySmack, 2024-09-19:
This did the trick....
I turned on the setting in the iobroker MQTT server:
[image]

In Home Assistant I defined the MQTT Number entity with the corresponding command_topic attribute:
```
  number:
    - name: "Control of AC Output Limit"
      unique_id: "YYYYYsetOutputLimit"
      state_topic: zendure-solarflow/0/XXXX/YYYY/control/setOutputLimit
      command_topic: zendure-solarflow/0/XXXX/YYYY/control/setOutputLimit/set
      value_template: >
        {{ int(value, 0) }}
      min: 0.0
      max: 800.0
      step: 1.0
      mode: "box"
      unit_of_measurement: "W"
      icon: "mdi:home-plus"
      device:
        name: "XXXX Hyper 2000"
        identifiers: "XXXXXXXX"
        manufacturer: "Zendure"
        model: "SolarFlow Hyper 2000 Controller"
```

Many thanks for your help.

Kind regards,
Torsten

closed 2024-09-21 by FlySmack

nograx, 2024-09-24:
Hi,

I'm trying to reproduce this to make an manual for other HA users. It's working to set the value from HA in ioBroker with your template. But my sensors in HA are not refreshing when an update in ioBroker is done. I can see the changes coming in with MQTT explorer. I've used your template above and added a sensor for solarInputPower. Both are only refreshed when HA or the MQTT Adapter in ioBroker is restartet. Any suggestions?

Kind regards,
Peter

FlySmack, 2024-09-25:
Hi @nograx ,

In the MQTT Server on the IOBroker, you have to turn on the "Send states" feature. Afterwards you receive updates every time something is changing on the zendure-solarflow adapter.
[image]

In Home Assistant, you have to ensure that the Mosquito Broker Add-On has the following customisation:
[image]

The following steps require SSH access to Home Assistant:
Afterwards I had created a configuration file in /root/share/mosquito:

cat zendure.conf
connection zendure-iobroker
address <ip/fqdn iobroker>:<port>
remote_username <iobroker mqtt user>
remote_password <ionbroker mqtt user>
remote_clientid hass-client
topic zendure-solarflow/# in 0
topic zendure-solarflow/0/XXXX/YYYY/control/# both 0

FlySmack, 2024-09-25:
To add the states and controls in Home Assistant, I did the following steps:

The make the handling of the MQTT entities much easier, I changed to configuration.yaml to be able to configure them in a separate file by adding the following line (all already existing MQTT entities in the configuration.yaml must be moved to the new file):

`mqtt: !include mqtt.yaml`

Now I created the file mqtt.yaml the same directory as the configuration.yaml (path is /root/config) and added all my MQTT entities, including the ones I already had previously configured in the configuration.yaml.

Attached the full mqtt.yaml with all the MQTT entities I use from the zendure-solarflow IOBroker adapter:
[mqtt.txt](https://github.com/user-attachments/files/17125115/mqtt.txt)

nograx, 2024-09-26:
Seems to be a little easier. I've created an wiki article about that: https://github.com/nograx/ioBroker.zendure-solarflow/wiki/ioBroker-Adapter-zum-Steuern-und-Auslesen-des-Solarflow-Ger%C3%A4tes-in-Home-Assistant-nutzen

Thanks a lot for your detailed information. The good thing is - after 2 years I can now control my blink cameras from ioBroker (there is no working ioBroker adapter - but an HA integration) :)

FlySmack, 2024-09-27:
The wiki article looks good. Just one remark about the MQTT step in Home Assistant:
In cases where users already having configured the MQTT add on as an MQTT Broker Server, your approach will not work and they will get the following error:
[image]

Reason is, that Home Assistant only supports one instance of MQTT. In such cases the users have to change the MQTT configuration the way I had posted above by configuring the MQTT Broker with a "bridge".

Regards,
Torsten