nograx/ioBroker.zendure-solarflow #66: Login works fine but [onReady] Found 0 SolarFlow device(s).

weggetor, 2024-09-29:
My Zendure Hyper 2000 works fine and is shown in the app. 

But within the adapter in the logs it shows after sucessful login: [onReady] Found 0 SolarFlow device(s).

What could be the problem? 
How can I check if I registered my Hyper2000 to Global and not to EU server. 
If I made it wrong (is registered on EU) - How can I change this ?

Please help!
Best , Torsten

FlySmack, 2024-10-17:
Hi @weggetor ,

The easiest way to test this is by using curl:
EU SERVER:
curl -i -X POST -H 'Content-Type: application/json' -d '{"snNumber": "<HYPER SN>", "account": "<ZENDURE APP ACCOUNT>"}' https://app.zendure.tech/eu/developer/api/apply

GLOBAL SERVER:
curl -i -X POST -H 'Content-Type: application/json' -d '{"snNumber": "<HYPER SN>", "account": "<ZENDURE APP ACCOUNT>"}' https://app.zendure.tech/v2/developer/api/apply

If the return is a JSON with code 200 and some data, then this is the registered region. Otherwise you get a JSON with code 400 and the message "Please bind the device first".

To change the region, you have to remove the Hyper in the Zendure App. Then start to add it again. One of the first pages on the registration process, it will ask you about the country (I don't speak about the country selection for the grid settings). Here ensure that you select "Global", instead the real country in Europe you live in. This will register it against the Global server.

Selecting any other country in Europe it will register it against the European servers.

Hopes this helps.
Regards,
Torsten

nograx, 2024-10-18:
Hi Torsten, 

I think I missed this one... did you get the adapter found your devices now? Or do you need further assistance?

Kind regards,
Peter

weggetor, 2024-10-25:
Hi and thanks to you both - I ve been in holidays, so sorry for late reaction. 

Yes, @FlySmack , your tips helped me. I removed my Hyper2000 first and then added it back again but this did not help. The question for which country I register is part of the user account, not of the device. 
After changing the ACCOUNT region to global and adding the device again it worked perfekt.

I also had to do this for my second account (which I use in iobroker) and add the shared device again.

closed 2024-10-25 by weggetor