nograx/ioBroker.zendure-solarflow #68: Update stable version in repo from 1.8.5 to 1.8.8

GermanBluefox, 2024-10-21:
# Think about update stable version to 1.8.8
**Version**: stable=**1.8.5** (55 days old) => latest=**1.8.8** (30 days old)
**Installs**: stable=**309** (61.31%), latest=**78** (15.48%), total=**504**

Click to use [developer portal](https://www.iobroker.dev/adapter/zendure-solarflow/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L3245)

**Do not close this issue manually as a new issue will be created if condition for update still exists.**

Please drop a comment if any reason exists which blocks updating to version 1.8.8 at this time.

Note: This is an automatically generated message and not personally authored by bluefox!
      @mcm1957 for evidence

mcm1957, 2024-11-02:
Is there any reason to delay updating this adapter at stable any longer?
Please drop a commit if you know any reason to refrain from updating.

**If there is no response until 16.11.2024 we could decide to update anyway**

nograx, 2024-11-02:
I‘m busy due to renovations at home. Will have a look at it this week.

nograx, 2024-11-08:
@mcm1957 How can I restart the process of https://github.com/ioBroker/ioBroker.repositories/pull/4201 ?

mcm1957, 2024-11-08:
What do you wa t to restart?
If tests are ok now I will chwxk next week after meeting @solingen.

nograx, 2024-11-08:
1.8.8 to stable repo. Was blocked due to nodejs 22…

closed 2024-11-13 by GermanBluefox