nograx/ioBroker.zendure-solarflow #70: Hyper 2000 incomplete topics ?

MaisonGF, 2024-10-25:
Here : https://github.com/reinhard-brandstaedter/solarflow-control/issues/272#issuecomment-2219808017

It seems we can get smartmeter and smartmode properties from our Zendure devices, it would be very interesting to have those two because there's no need to do a complex blockly automation if the logic is ondevice. + we could use any sensor to give it to the smartPower property and control the logic (CT matching on day by changing smartMode, charging at night changing it again).

[image]

Thank you for all the work !

+ by the way it a complete blockly automation for the Hyper 2000, it's very difficult to find for non deutch people :), it's here, feel free to add it as example (here i got only ac coupled Hyper 2000, smart CT with slower discharge under 20%, and charge at night with cheap energy to have some in the morning, waiting for the sun) : 
[Hyper2000_Automation1(1).xml.txt](https://github.com/user-attachments/files/17518108/Hyper2000_Automation1.1.xml.txt)

Quickiman, 2025-04-03:
Is it possible to create a functiinality like the smart CT-Mode? If yes is there a chance that you will implement it?

nograx, 2026-08-18:
Finally about 2 years later the adapter (v 5.0.0) will now expose all states provided by the device 💯

closed 2026-08-18 by nograx