nograx/ioBroker.zendure-solarflow #72: MQTT Access - writing a property

jebos, 2024-11-10:
Hello,
not a direct bug. I wanted to understand how you write properties. I followed your code and ended up here:

```
const topic = `iot/${productKey}/${deviceKey}/properties/write`;
const inputLimitContent = { properties: { inputLimit: limit } };
adapter.mqttClient?.publish(topic, JSON.stringify(inputLimitContent));
```

I use MQTT Explorer on my PC, log into my Zendur Account and get live-feed data. 
However - what ever I publish to this `iot/${productKey}/${deviceKey}/properties/write` topic (I changed my product and device key) it has no effect. - I have a hyper 2000. All modes are disabled / off. 

AC charging is enabled in the App.

What do I overlook here?

jebos, 2024-11-11:
Alright, got it. Thanks for the great work with this adapter.

closed 2024-11-11 by jebos

nograx, 2024-11-11:
What is the solution?

jebos, 2024-11-11:
Following the entire code of yours and extracting whats needed.

First; Get the access token.

With it comes the iot access data. And one can request the devicelist.

Second; Extracting the required fields from the responses. 

Third; Setting clientid, user, password,.. as you do and connect to the mqtt server.

Forth; Publish to the topics.

cmuster, 2024-12-07:
@jebos I'm also trying to setup a mqtt connection via python or mqtt explorer the same way the ioBroker adapter does.
I've successfully retrieved the device list with the received token, but when I setup the mqtt connection with the broker url `"mq.zen-iot.com",` the received token as clientId, `"zenApp"` as username and the hardcoded pw:

https://github.com/nograx/ioBroker.zendure-solarflow/blob/be7f1de2a83c97342d108fe11e03aacc791b5c69/src/constants/paths.ts#L18

the connection is refused with bad username or pw..

@nograx  sorry for posting in this already resolved issue, but is there anything I'm missing here?

nograx, 2024-12-07:
The password in this file is encrypted, you have to decrypt it first with base64.

cmuster, 2024-12-07:
Thanks for the quick response! Connection is working now 👍