nograx/ioBroker.zendure-solarflow #75: iobroker logs shows State "zendure-solarflow.0.xxx.yyyy.control.hubState" has no existing object, this might lead to an error in future versions

FlySmack, 2024-12-06:
**Describe the bug**  
After restart of the Zendure instance, the following warning is raised in the iobroker logs:
State "zendure-solarflow.0.xxxxx.yyyyyy.control.hubState" has no existing object, this might lead to an error in future versions

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'Instances'
2. Click on 'Restart' button for the Zendure instance
3. Go to 'Logs'
4. See error

**Expected behavior**  
N/A

**Screenshots & Logfiles**  
State "zendure-solarflow.0.xxxxx.yyyyyy.control.hubState" has no existing object, this might lead to an error in future versions

**Versions:**  
 - Adapter version: 1.10.0
 - JS-Controller version: 7.0.2
 - Node version: v20.18.1
 - Operating system: docker (official image - v10.0.0)

**Additional context**  
Looks like something which might be no longer supported in future iobroker versions?!

nograx, 2024-12-06:
Hi, do you use smart plugs? Possibly this issue comes from them...

FlySmack, 2024-12-06:
Hi, no. I don’t use smart plugs, or the ct mode.
I’m not 100% sure, when these messages started the first time. A Hyper2k firmware update and Zendure–solarflow update happened in a very short period of time. Shall I use an Mqtt client to check for something specific reported by the Zendure  Mqtt servers?

nograx, 2024-12-10:
Can you show me the xxxxx path in your state above? It's the product key which shows me the exact product which throws this error. I DON'T need the yyyyyy part which is your serial number of the device.

bersu72, 2024-12-12:
Hi
hab die 1.10.0 Version installiert und bekomm die selbe Fehlermeldung
State "zendure-solarflow.1.A8yh63.0A848C52.control.hubState" has no existing object, this might lead to an error in future version
State "zendure-solarflow.1.gDa3tb.EHV1FAq3.control.hubState" has no existing object, this might lead to an error in future version

Meldung1 stamm vom Hub2000   Meldung2 vom Hyper

FlySmack, 2025-01-14:
The error message no longer appears on my iobroker. Not sure what exactly fixed it, but must have been one of the following updates (or a combination of some):
zendure-solarflow updated to: 1.10.1
admin GUI updated to: 7.4.12
js-controller updated to: 7.0.6

bersu72, 2025-01-14:
Selbe verhalten bei mir, seit Version 1.10.1 ist der Fehler weg

closed 2025-01-27 by nograx