nograx/ioBroker.zendure-solarflow #80: Problem changes energy plan

technoman63, 2025-02-23:
**Describe the bug**  
Problem with energy plan changes with Hyper 2000.

**Expected behavior** 
 
It is impossible to control the energy plans, when you choose an energy plan or when you want to deactivate it, the action is done but do not hold it, it lasts about a few seconds and after that it returns to the current plan.

**Versions:**  
 - Adapter version: 1.11.0

intruder7777, 2025-05-16:
same here. SetOutputLimit set to 150 . After about 2 seconds it is changing back to 0

FlySmack, 2025-07-20:
I also have the same issue with my Hyper2k. Setting the energy plan to nothing will be automatically switched back to the last active plan. Turning the energy plan off with the Zendure App gets permanent and is reflected in the I/O Broker adapter. Is there anything I can collect with an MQTT tool from the Zendure Cloud, when switching the energy plan through I/O Broker and Zendure App to analyse what is happening? For me it looks like that the Zendure App is doing something additionally to just changing the energy plan.

Thanks and regards,
FlySmack