nograx/ioBroker.zendure-solarflow #81: Local Control

jafrka, 2025-02-25:
It is possible to change the MQTT settings of the Hub etc. to a local MQTT server, see [https://github.com/reinhard-brandstaedter/solarflow-bt-manager]. Would it be possible to add such setting to this adapter? I.e. a third option for the connection (EU, Global, **Local**). For the Local option the mqtt server url needs to be adjustable. Also, the token handling is not necessary for this part.

andig, 2025-03-01:
Local control has been mentioned to be available for first half of 2025.

nograx, 2025-03-13:
Maybe there is a solution for this. My problem is that currently I don't have a test device for this.

andig, 2025-03-13:
What solution? https://github.com/dav1dBoy/zenSDK does not support the Hub.

andig, 2025-03-13:
My use case: what I'd love to do is simulate a Shelly 3EM- either writing grid surplus to the Zendure or providing a JSON api that looks _like_ a Shelly and have the Zendure detect that.

nograx, 2025-03-18:
It's now possible. You have 3 methods to get the Solarflow device "offline". solarflow-bluetooth-manager from Reinhard Brandstätter, my Windows tool ([zendure-cloud-disconnector](https://github.com/nograx/zendure-cloud-disconnector)) or rerouting DNS traffic from [mq.zen-iot.com](http://mq.zen-iot.com/) to your own network. You need a MQTT instance on port 1883 without authentication.

Then use version 1.12.1 of the adapter and select "local" as server. The ioBroker adapter will then connect to your mqtt instance and works as before. It's some kind of beta - but first responses are that it is working good.

closed 2025-03-18 by nograx