nograx/ioBroker.zendure-solarflow #84: Get total energy producation and other stats

philiip11, 2025-03-22:
Hi 👋 

The Zendure App allows to see historical stats of ones devices, like how much energy the solar panels have produced in total etc.

Does anyone have an idea how the app fetches that information?
Are there other MQTT-topics one has to listen to?

Would be nice if this information is available via ioBroker.

philiip11, 2025-03-23:
I did some reverse engineering using an old Android Tablet and mitmproxy:
The app makes two REST-Calls for the stats.

`POST https://app.zendure.tech/eu/tdengine/device/solarFlow/energy` 
This returns stats for energy consumption, storage and production.

`POST https://app.zendure.tech/eu/tdengine/device/solarFlow/electric` 
This returns stats for battery percentage (I think average per timeframe).

Replace `eu` with `v2` if you're not in Europe.

Use the same headers one uses for the `productModule/device/queryDeviceListByConsumerId`-Endpoint.

The request-body for both endpoints looks like this:

```
{
    "deviceId": 12345, 
    "beginDate": "2025-03-23", 
    "endDate": "2025-03-23",
    "aceId": "",
    "zone": "Europe/Berlin",
    "type": 0
}
```

`deviceId` is the id returned by `queryDeviceListByConsumerId`.
`beginDate` and `endDate` are YYYY-MM-DD dates for timespans or an empty string if you want total data.
`aceId` is always an empty string for me, I'm using a Hyper 2000. Maybe it is used for other hardware.
`zone` is your timezone.
`type` is a number from 0 to 4 or an empty string: 

-   `0` returns one datapoint per minute (only available for single days
-   `1` returns one datapoint per day
-    `2` returns one datapoint per week
-    `3` returns one datapoint per month
-    `4` returns one datapoint per year
-    `""` if you don't set `beginDate` and `endDate`, you can leave this an empty string and one datepoint per month.

The response looks like this
```
{
    "code": 200,
    "success": true,
    "data": {
        "energyVos": [],
        "data": {
            "1": [
                {
                    "dateTime": "2024-12",
                    "data": 4953.14,
                    "source": 1,
                    "sortNo": 0
                },
                {
                    "dateTime": "2025-01",
                    "data": 20902.21,
                    "source": 1,
                    "sortNo": 1
                },
                {
                    "dateTime": "2025-02",
                    "data": 38360.36,
                    "source": 1,
                    "sortNo": 2
                },
                {
                    "dateTime": "2025-03",
                    "data": 80532.77,
                    "source": 1,
[... 166 lines cut ...]
                {
                    "dateTime": "",
                    "data": 0,
                    "source": 11,
                    "sortNo": -1
                }
            ]
        },
        "type": 3,
        "solar": 239056.23,
        "home": 377498.01,
        "batteryInput": 327912.43,
        "batteryOutput": 382811.29,
        "outputToInverse": 0,
        "bindDeviceInput": 0.0,
        "outputToBindDevice": 0.0,
        "gridInputTotal": 214285.16,
        "acOutputTotal": 0.0,
        "dcOutputTotal": 0.0
    },
    "msg": "Erfolgreiche Operation"
}
```

That should be enough info to implement it yourself ;-)

Quickiman, 2025-03-24:
Hey guys. I m currently on 1.11.0 stable which I installed via iobroker. How can I install the latest 1.12.7? Sry for offtopic