nograx/ioBroker.zendure-solarflow #88: Vorschläge wegen offline Mode

funflo, 2025-04-29:
Ich habe es gestern geschafft, den Hub mit dem Disconnector offline einzubinden - läuft wunderbar. Ich wünschte nur, ich könnte zum einen den Hub-State (Verhalten bei erreichter Endladegrenze, ohne dass der Strom bekommt) steuern. Zum anderen, wenn es möglich ist, die IP-Adresse auszulesen wie in der APP.

nograx, 2025-04-29:
Kannst du mir zur Sicherheit mal den ProductKey (nicht DeviceKey) reinschreiben falls es nicht "A8yh63" ist?

funflo, 2025-04-29:
Jo:
1. A8yh63 (aktuell hub2000)
2. 73bkTV  (ist aktuell aber nicht mehr angeschlossen habe nur die dp noch in iobroker hub1200)

Bei beiden in der aktuellen version kein Hubstate in Controll.

closed 2025-04-30 by nograx