nograx/ioBroker.zendure-solarflow #94: Feature Request: one object for battery charge/discharge

andiweli, 2025-05-17:
Hi

The battery of the SolarFlow can either be charged OR discharged.
I am using the "power-distribution-card.js" on Lovelace - it can't use two different states for one visualisation. So currently I am using "outputPackPower" and packInputPower" and it shows two different visualisations.

But it can use positive and negative values and show the flow (from or to the battery).
In the attached screenshot the bottom one on the left side shows either blue value and arrows of I am taking power from the grid or white values and arrows when I am giving power to the grid. I have one value for it that's either positive (taking from grid) or negative (giving to grid).

[image]

Would it be possible to have a "calculation" with both values? Either it's a positive value (if charging) or a negative one (if battery is depleting). So it would be possible to have only one visualisation with arrows either to the left (charging) or right (depleting).

Kind regards

Daisyaudia, 2025-06-17:
> There is Lovelace as UI for ioBroker.
👍

If the object will not implemented in the adapter, here my quick n dirty solution.
Maybe it helps as a simple template.

-----------------

trigger:
If objects
outputPackPower
or packInputPower
or packState
have been changed
do:

If the value of the object ID packState != 'Idle'
and the value of the object ID outputPackPower > 0
update selfMadeDataPoint with the value of the object ID outputPackPower

If the value of the object ID packState != 'Idle'
and the value of the object ID packInputPower = 0
update selfMadeDataPoint with the value of the object ID packInputPower * -1

If the value of the object ID packState = 'Idle'
or
the value of the object ID packInputPower = 0
and
the value of the object ID outputPackPower = 0
update selfMadeDataPoint with the value 0

----------------------

folder/Datapoint: 0_userdata.0.Zendure.selfMadeDataPoint
productId: gDa3tb (Hyper2000)
deviceID: XxxXXxxX

js:

`on({ id: [].concat(['zendure-solarflow.0.gDa3tb.XxxXXxxX.outputPackPower']).concat(['zendure-solarflow.0.gDa3tb.XxxXXxxX.packInputPower']).concat(['zendure-solarflow.0.gDa3tb.XxxXXxxX.packState']), change: 'ne' }, async (obj) => {
  if ((getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.packState').val != 'Idle') && (getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.outputPackPower').val > 0)) {
    setState('0_userdata.0.Zendure.selfMadeDataPoint' /* selfMadeDataPoint */, getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.outputPackPower').val, true);
  }
  if ((getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.packState').val != 'Idle') && (getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.packInputPower').val > 0)) {
    setState('0_userdata.0.Zendure.selfMadeDataPoint' /* selfMadeDataPoint */, (getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.packInputPower').val * -1), true);
  }
  if ((getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.packState').val == 'Idle') || ((getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.outputPackPower').val == 0) && (getState('zendure-solarflow.0.gDa3tb.XxxXXxxX.packInputPower').val == 0))) {
    setState('0_userdata.0.Zendure.selfMadeDataPoint' /* selfMadeDataPoint */, 0, true);
  }
});`

nograx, 2025-07-11:
Starting with version 1.15.0 there is "packPower". There is a little Bug so the value is not set to 0 if no input/output, this is fixed in version 1.15.1.

closed 2025-07-11 by nograx