nograx/ioBroker.zendure-solarflow #105: SolarFlow 800 to ioBroker

Coyotex1, 2025-06-29:
Hi,

Its not a issue but i dont know how i add the new SolarFlow 800 to the ioBroker, i dont find the option share the device, like the Hub 2000 and Ace 1500, how did it work or get it to the ioBroker.

Can someone give me a hint or how i get it to work?

thank you

nograx, 2025-06-30:
I think starting with Solarflow 2400 AC and Solarflow 800 it's not possible to share devices anymore.

Coyotex1, 2025-07-01:
hi,

thanks for your fast answer?

when its not possible, how get it the Solarfow 800 in the second and third account? because it not showing, the solarflow and smartmeter 3ct is only in the primary app at phone? Because in the Zendure Adapter changelog at ioBroker is written compatible with Solarflow 800

thank you

nograx, 2025-07-11:
You have to use your primary account, sharing to a second one doesn't work anymore.

closed 2025-07-11 by nograx