nograx/ioBroker.zendure-solarflow #108: Frage calculations

funflo, 2025-07-04:
Moin,
ich habe da doch nochmal zwei fragen.
es gibt ja den wert soc ich gehe mal davon aus das es sich um den wert handelt der nutzbar ist (eigentlich das was man sich wünschen würde). Wie verhält sich dieses wenn man z.b. den wert der Zendure selber auf 50 % stellt (Winter) verändert sich dann dieser auch?

Desweiteren wie verhält sich das mit dem wert energyWhMax? Hier ich musste hier einen wert manuel eintragen wird dieser auch automatisch korrigiert wenn ich diesen zu hoch oder zu niedrige eingestellt habe? 

Danke schon mal für die antwort :)

CBRocker, 2026-08-27:
Hm, 
die Fragen sind etwas unklar.

Frage 1:
Beim Akku gibt es im Prinzip 3 Werte:

SoC: aktuelle Kapazität des Akkus
SoC min: minimale Schwelle, bis zu der entladen werden soll
SoC max: maximale Schwelle, bis zu der geladen werden soll

Was genau meinst du mit Frage 1?

Frage2:
Was meinst du mit dem Wert energyWhMax? Wo hast du was für einen Wert manuell eintragen müssen?

Grüße Chris

funflo, 2026-08-27:
Moin, sorry ich dachte dieser Artikel ist nicht mehr aktiv. Hat sich mittlerweile erledigt, Danke trotzdem das nach so langer Zeit eine Antwort kommt.

closed 2026-08-27 by funflo